gnea/grbl-Mega #157: How to Compile for Arduino Mega and Ramps 1.4

Technican, 2022-04-10:
I have tried to compile it for the Arduino Mega 2560 and Ramps 1.4 Board.
I changed the in the Config.h:

#include "grbl.h" // For Arduino IDE compatibility.

// Define CPU pin map and default settings.
// NOTE: OEMs can avoid the need to maintain/update the defaults.h and cpu_map.h files and use only
// one configuration file by placing their specific defaults and pin map at the bottom of this file.
// If doing so, simply comment out these two defines and see instructions below.
#define DEFAULTS_GENERIC
#define CPU_MAP_2560_INITIAL

// To use with RAMPS 1.4 Board, comment out the above defines and uncomment the next two defines
//#define DEFAULTS_RAMPS_BOARD
#define CPU_MAP_2560_RAMPS_BOARD

Is that correct?
After that , with make in the Terminal:

avr-gcc -Wall -Os -DF_CPU=16000000L -mmcu=atmega2560 -I. -ffunction-sections -flto -MMD -MP -c grbl/main.c -o build/main.o
In file included from grbl/grbl.h:47,
                 from grbl/main.c:22:
grbl/cpu_map.h:160: warning: "STEP_DDR" redefined
  160 |   #define STEP_DDR(i) _DDR(STEP_PORT_##i)
      | 
In file included from grbl/grbl.h:47,
                 from grbl/main.c:22:
grbl/cpu_map.h:36: note: this is the location of the previous definition
   36 |   #define STEP_DDR      DDRA
      | 
In file included from grbl/grbl.h:47,
                 from grbl/main.c:22:
grbl/cpu_map.h:162: warning: "STEP_PORT" redefined
  162 |   #define STEP_PORT(i) _STEP_PORT(i)
      | 
In file included from grbl/grbl.h:47,
                 from grbl/main.c:22:
grbl/cpu_map.h:37: note: this is the location of the previous definition
   37 |   #define STEP_PORT     PORTA
      | 
In file included from grbl/grbl.h:47,
                 from grbl/main.c:22:
grbl/cpu_map.h:163: warning: "STEP_PIN" redefined
  163 |   #define STEP_PIN(i) _PIN(STEP_PORT_##i)
      | 
In file included from grbl/grbl.h:47,
                 from grbl/main.c:22:
grbl/cpu_map.h:38: note: this is the location of the previous definition
   38 |   #define STEP_PIN      PINA
      | 
In file included from grbl/grbl.h:47,
                 from grbl/main.c:22:
grbl/cpu_map.h:174: warning: "DIRECTION_DDR" redefined
  174 |   #define DIRECTION_DDR(i) _DDR(DIRECTION_PORT_##i)
      | 
In file included from grbl/grbl.h:47,
                 from grbl/main.c:22:
grbl/cpu_map.h:45: note: this is the location of the previous definition
   45 |   #define DIRECTION_DDR     DDRC
      | 

and so on. Any ideas?

fra589, 2022-04-10:
Hi @Technican,  
  
Your config.h update is not correct. To use an Arduino Mega with RAMPS 1.6, you need to comment the two first definitions and uncomment the two other like this:  
```
// Define CPU pin map and default settings.
// NOTE: OEMs can avoid the need to maintain/update the defaults.h and cpu_map.h files and use only
// one configuration file by placing their specific defaults and pin map at the bottom of this file.
// If doing so, simply comment out these two defines and see instructions below.
//#define DEFAULTS_GENERIC
//#define CPU_MAP_2560_INITIAL

// To use with RAMPS 1.4 Board, comment out the above defines and uncomment the next two defines
#define DEFAULTS_RAMPS_BOARD
#define CPU_MAP_2560_RAMPS_BOARD
```

Then, if you have other compile error message, please, tell us what is your system, Windows or Linux, which version, which compiler and version, and any other information which can help us to help you...
  
@++;  
Gauthier.

Technican, 2022-04-10:
Hi Gauthier,

Thanx for the fast answer. The changes from you helps me. I am using macOS 12 , avr-gcc from Homebrew  , Version 9.3.0.3.
Only a warning appears during the Compilation, is this a problem?

Great Rewards from Germany ,
Sven

grbl/eeprom.c: In function 'memcpy_to_eeprom_with_checksum':
grbl/eeprom.c:133:26: warning: '<<' in boolean context, did you mean '' ? [-Wint-in-bool-context]
  133 |     checksum = (checksum << 1) || (checksum >> 7);
      |                ~~~~~~~~~~^~~~~
grbl/eeprom.c: In function 'memcpy_from_eeprom_with_checksum':
grbl/eeprom.c:144:26: warning: '<<' in boolean context, did you mean '' ? [-Wint-in-bool-context]
  144 |     checksum = (checksum << 1) || (checksum >> 7);
      |                ~~~~~~~~~~^~~~~

fra589, 2022-04-10:
I work under Linux Debian, and the current version in Debian stable is GCC 5.4.0.
Under this version, I don't have those warning. But GCC version 9 is more finicky and tell many more warning.
Regarding the source code, I don't think this will be a problem, just a missing cast between integer and boolean and should work fine.
The warning is in the code that saves and restores Grbl settings to EEPROM. If Grbl settings save and return correctly after powering down, it's working. :-)
@++;  
Gauthier.

drf5n, 2022-04-11:
> grbl/eeprom.c: In function 'memcpy_to_eeprom_with_checksum': grbl/eeprom.c:133:26: warning: '<<' in boolean context, did you mean '' ? [-Wint-in-bool-context] 133 | checksum = (checksum << 1) || [...]

That looks like it was a typo -- it squashes the checksum down to a single bit of information, when it looks like it intended to roll it right one bit, so the checksum ends up either zero or one larger than the last character:

https://github.com/gnea/grbl-Mega/blob/df87b36f48b90930714accf03b411b60d5d948e5/grbl/eeprom.c#L130-L149

Maybe it was supposed to be  

    checksum = (checksum << 1) | (checksum >> 7);

fra589, 2022-04-11:
Very old subject... Already discussed in 2017 here : 
https://github.com/grbl/grbl/issues/1249
@++;
Gauthier.