gnea/grbl-Mega #164: Can't get any output in serial monitor

Lucas-Kaiser, 2022-07-20:
Hello, 
I'm trying to control a custom made 3D-Printer with an Arduino Mega and GRBL. Currently I'm using an Arduino Uno for that and everything works. 

For the Mega I downloaded the grbl-Mega and uploaded it to the Arduino Mega with the Arduino IDE 1.8.19. The upload works but when I open the serial monitor nothing shows up (it's just empty) and I can't get any output (trying to type "$").

I wrote a small programm to test the serial connection and it works just fine. 

Do I have to change any settings within the grbl-Mega to make it work?

Can anyone help me with this issue?
Thanks.

fra589, 2022-07-20:
Hi @tekkzLK 

The default communication speed of grbl-Mega is 115200 bps.
Do you configure Arduino serial monitor to this speed ?
The Arduino default serial monitor speed is 9600 bps.

@++;
Gauthier.

tekkzLK, 2022-07-20:
Hi @fra589 ,
thanks for the quick response.
Yes, the speed is configured to 115200 bps.

> Hi @tekkzLK
> 
> The default communication speed of grbl-Mega is 115200 bps. Do you configure Arduino serial monitor to this speed ? The Arduino default serial monitor speed is 9600 bps.
> 
> @++; Gauthier.

tekkzLK, 2022-07-20:
Hi @fra589,
thanks for the quick response.
Yes, the speed is set to 115200 bps.

closed 2022-07-20 by tekkzLK