gnea/grbl-Mega #166: Sg90 on mks base 1.6 with grbl

alexc57, 2022-11-10:
Impossible to do moving the servo with many different firmware. Anybody can help. Maybe a bad configuration because I have plus the 3 pin in the corner (servo pin). Thks

fra589, 2022-11-10:
Grbl doesn't have code to drive servo.
You need to use another firmware like Marlin to do this.
@++;
Gauthier.

alexc57, 2022-11-10:
It s strange I see many informations about that.

fra589, 2022-11-10:
Some modified versions exist to use servo (https://github.com/robottini/grbl-servo, https://github.com/cprezzi/grbl-servo, ...), but they are no servo function in the original (this one) Grbl version.

Another infos here: 
- https://www.instructables.com/How-to-Control-a-Servo-Using-GRBL/
- https://www.google.com/search?q=grbl+servo

@++;
Gauthier.

alexc57, 2022-11-11:
I have try this also, google is my friend...
but all it s for ATmega328p, but not for my board.
if you had another idea ;)

fra589, 2022-11-11:
I don't know if they work, but try this: 
https://github.com/botio/grbl-Mega-servo
Or this:
https://github.com/alnwlsn/grbl-Mega-5X-alnwlsn

@++;
Gauthier.

alexc57, 2022-11-11:
thanks a lot but doesn't work, it is not the same board

fra589, 2022-11-11:
You can remap the IO pins to make the soft suitable to your board...
https://github.com/fra589/grbl-Mega-5X/wiki/Pinout-mapping-in-cpu_map.h

alexc57, 2022-11-11:
Thanks a lot for that. In my mks base I don't have pin (I/o) for spindle!
To you think a can attribute another pin when you explain? Like (OC1A/PCINT5)PB5? For example ? Thks