gnea/grbl-Mega #173: X Axis on E0 Axis Pins?? MKS Gen_L V1.0 board

Horsboll, 2024-03-23:
I have to throw in the towel. I'm stuck..
I've been trying for a long time now and I can't get it to work. I am in the process of converting my 6040 CO1 laser to GRBL on an MKS Gen_L v1 board and everything works except the X axis. I'm sure the X axis is damaged.
I need 3 axis,  X, Y and Z and there are 5 on the board, so I think that I shold be able to change the X axis to use E0 pins? I´m using external drivers, so I just have to change "E0 STEP" to "X STEP" and "E0= DIRECTION" change to "X DIRECTION". But I simply cannot figure out how.
I'm just not good enough to code yet to figure it out.

The only change I have made is to switch to the Ramps board, I have also changed the Spindle pin to one of the servo pins with which I make PWM 5-0 volts for the laser tube

Horsboll, 2024-03-24:
I have solved the problem my self
I couldn't wrap my head around how I could determine which pins it has on X step and dir and on E0 step and dir and then have them swapped around.. I succeed by finding a GRBL that is set up to 5 akas and then went through what kind of pins it calls for the E0 axis and put them into the X axis on my own cpu_map file.. and I just went out and tried it out and now run my laser in all 3 axes ..!!

 #define STEP_PORT_0 F  **// F-->A**
  #define STEP_PORT_1 F
  #define STEP_PORT_2 L
  #define STEP_BIT_0 0  // X Step - Pin A0 -  **_0 0 --> _0 4 = Pin D26**
  #define STEP_BIT_1 6  // Y Step - Pin A6
  #define STEP_BIT_2 3  // Z Step - Pin D46
  #define _STEP_BIT(i) STEP_BIT_##i
  #define STEP_BIT(i) _STEP_BIT(i)
  #define STEP_DDR(i) _DDR(STEP_PORT_##i)
  #define _STEP_PORT(i) _PORT(STEP_PORT_##i)
  #define STEP_PORT(i) _STEP_PORT(i)
  #define STEP_PIN(i) _PIN(STEP_PORT_##i)

  // Define step direction output pins.
  #define DIRECTION_PORT_0 F **// F-->A**
  #define DIRECTION_PORT_1 F
  #define DIRECTION_PORT_2 L
  #define DIRECTION_BIT_0 1 // X Dir - Pin A1 - **_0 1 --> _06 Pin D28**
  #define DIRECTION_BIT_1 7 // Y Dir - Pin A7
  #define DIRECTION_BIT_2 1 // Z Dir - Pin D48
  #define _DIRECTION_BIT(i) DIRECTION_BIT_##i
  #define DIRECTION_BIT(i) _DIRECTION_BIT(i)
  #define DIRECTION_DDR(i) _DDR(DIRECTION_PORT_##i)
  #define _DIRECTION_PORT(i) _PORT(DIRECTION_PORT_##i)
  #define DIRECTION_PORT(i) _DIRECTION_PORT(i)
  #define DIRECTION_PIN(i) _PIN(DIRECTION_PORT_##i)

  // Define stepper driver enable/disable output pin.
  #define STEPPER_DISABLE_PORT_0 D  **//D-->A**
  #define STEPPER_DISABLE_PORT_1 F
  #define STEPPER_DISABLE_PORT_2 K
  #define STEPPER_DISABLE_BIT_0 7 // X Enable - Pin D38 - **_0 7 --> _02 Pin D24**
  #define STEPPER_DISABLE_BIT_1 2 // Y Enable - Pin A2
  #define STEPPER_DISABLE_BIT_2 0 // Z Enable - Pin A8
  #define STEPPER_DISABLE_BIT(i) STEPPER_DISABLE_BIT_##i
  #define STEPPER_DISABLE_DDR(i) _DDR(STEPPER_DISABLE_PORT_##i)
  #define STEPPER_DISABLE_PORT(i) _PORT(STEPPER_DISABLE_PORT_##i)
  #define STEPPER_DISABLE_PIN(i) _PIN(STEPPER_DISABLE_PORT_##i)

fra589, 2024-03-26:
Hi @Horsboll,  
  
The pinout mapping in cpu_map.h is describe here : https://github.com/fra589/grbl-Mega-5X/wiki/Pinout-mapping-in-cpu_map.h

@++;
Gauthier.

Horsboll, 2024-03-26:
Thanks it would have been helpful 👍🏻
But I figured it out already by thinking for a loooong time 🤣