gnea/grbl-Mega #174: Pins don't align with #define statements

ccwtruck, 2024-03-28:
I am attempting to install GRBL-Mega 1.1g (the current release) on an ATmega2560R3, and try as i might, I get no motion on Y & Z axis. According to the cpu_map.h, I have the stepper drivers connected properly:
`#define X_STEP_BIT    2 // MEGA2560 Digital Pin 24
#define Y_STEP_BIT    3 // MEGA2560 Digital Pin 25
#define Z_STEP_BIT    4 // MEGA2560 Digital Pin 26

// Define step direction output pins. NOTE: All direction pins must be on the same port.
#define X_DIRECTION_BIT   7 // MEGA2560 Digital Pin 30
#define Y_DIRECTION_BIT   6 // MEGA2560 Digital Pin 31
#define Z_DIRECTION_BIT   5 // MEGA2560 Digital Pin 32
`
I have flashed with your .hex file several times, and compiled/ uploaded from source several times -- still, the Y and Z axis's show no sign of life.
This is on a CNC plasma table that worked well on the Uno with v1.1h.
I have double, triple & quadruple-checked my wiring -- it is correct; after all, i'm just upgrading the system board, and not re-wiring any more than that requires.
Please, can you tell me exactly which physical pins do my stepper controllers need to connect to?
Thank you in advance for any assistance!!!

fra589, 2024-03-28:
Hi @ccwtruck  

Perhaps thie doc: https://github.com/fra589/grbl-Mega-5X/wiki/Pinout-mapping-in-cpu_map.h will help you to find which pins are used...  
  
@++;  
Gauthier.

ccwtruck, 2024-03-28:
Thank you for that; I've at least verified that my steppers are connected to the right pins. I am now wondering if my Y & Z steppers have taken a dump. They still light up, but with the connections being correct, that's the next logical link in the chain.

ccwtruck, 2024-05-17:
After temporarily abandoning my Mega2560R3 project, I picked it back up the other day. I re-flashed the Mega with @chamnit's latest version and now I have proper motion on both X & Z axis's. Y axis is still completely unresponsive. As a test, I initiated a long, slow G1 move on Y axis while I used a logic probe to try to determine if there was any output on pins 25 & 31 (25=STEP, 31=DIR).
There is a VERY STRONG NON-PULSING signal on pin 31 but nothing on pin 25 at all. When I switch the move to X axis, I get no signal on pin 30 and a real nice pulse signal on pin 24 -- exactly what I had expected, since the X axis is moving slowly. Z axis shows the same behavior as X axis,
It is obvious to me that there is an incorrect pin assignment for Y axis -- pins 25 and/ or 31 are NOT where they are assigned. I cannot confirm it with the tools I have, but it's as if a 5v voltage is being output at pin 31 -- the signal is that strong & steady.
Does anyone have any idea what is causing this other than a incorrect assignment, either in the #defines, or within the program flow?

closed 2024-06-14 by ccwtruck